PiNet/PiNet #158: PiNet creates user account

squadramunter, 2019-04-05:
Hello I want to disable the user's added as pupil in my login screen from server side. I know there is a way to hide a user from the login screen but don't know exactly how to do it.

I want to list teachers in the login screen from the server but pupil is to much. My login list is so big now.

[image]

Can this be solved in a feature update?

squadramunter, 2019-04-05:
Seems like you can add this peace of code to hide a user from login screen.

/var/lib/AccountsService/users/user1
and place this code in the created user file. 

[User]
SystemAccount=true

gbaman, 2019-04-05:
Although perhaps a tad annoying, this is a pretty low priority bug to fix in PiNet. In general most folks managing PiNet don't use the Ubuntu desktop to do so, instead opting to SSH from a Pi or simply use the application on the Pi desktop of any teacher account.

There is potentially unexpected consequences of the above suggested fix and right now I am not going to have the time to properly dig into it with the many other larger issues with PiNet :)

As such, this is going to go under the won't fix category at least for the foreseeable future.

closed 2019-04-05 by gbaman

squadramunter, 2019-04-06:
http://pinet.org.uk/articles/installation/installing-ubuntu.html 

This is a Desktop edition not a Server edition right?

http://pinet.org.uk/articles/installation/installing-PiNet.html 

Why not updating the manual to not confuse every user.

I have not idea how I can reach the user folder from a raspberry client because it is not showing up there.