PiNet/PiNet #160: JamHat not supported yet?

birv2, 2019-05-15:
Just got one of these new JamHats, supposedly supported on gpiozero 1.5, but not recognized on my current install of PiNet (1.2.10). Suggestions?

gbaman, 2019-05-18:
That is odd. Can you @birv2 try running the below on a Pi running on PiNet in LXTerminal and post back the resulting version?
`pip3 list | grep gpiozero`

birv2, 2019-05-20:
It said " gpiozero(1.4.0) you are using pip verions 9.0.1, however, version 19.1.1 is available. You should consider upgrading via the pip install --upgrade pip command."
I'm pretty sure it's that gpiozero 1.4 doesn't support the JamHat. How to upgrade it on the server, though?

gbaman, 2019-05-20:
It should be included in the "Update all" Option in PiNet. That includes installing updates with pip from PyPi. Worth giving that a go, see if it pulls down the most recent version.

birv2, 2019-05-21:
Tried that but pip and pip3 wouldn't update because of dependencies in Ubuntu. Truth be told, I've been holding off on Ubuntu updates for fear of breaking something in PiNet. I will do that in a week, when classes are over.

gbaman, 2019-05-29:
Can confirm after a remote session, that this was actually an unrelated bug blocking the compressed image being updated. It is now resolved.

closed 2019-05-29 by gbaman