PiNet/PiNet #163: Change display setting

seksitha, 2019-08-15:
I'm testing Pinet. And it boots successfully. 
I have display setting issue where the screen resolution is 800x600 or something. the text is too big.

How can I change that from the server or client?

seksitha, 2019-08-15:
So I found the solution going through the doc!

In the server go to Pi bash. \
`$ sudo ltsp-chroot --arch armhf`
edit config.txt
`nano /boot/config.txt`
add this line
`hdmi_group=16 # this will force 1024x768 resolution`
[More about config here](https://www.raspberrypi.org/documentation/configuration/config-txt/video.md)

closed 2019-08-15 by seksitha

gbaman, 2019-08-15:
@seksitha so be warned, that if you edit the config.txt file directly, it will get overwritten at a later date when the system updates the boot files remotely.
There is actually a built in process for editing the config.txt file, right from within PiNet (that pushes the edits out to all PiNet Pis).
See - http://pinet.org.uk/articles/advanced/custom-config-file.html

seksitha, 2019-08-15:
Yes, I try that already but somehow it does not work. I might miss something. will try again. 
Thank you for that.

seksitha, 2019-08-15:
@gbaman  I have another thing to fix. The [known issue](https://raspberrypi.stackexchange.com/questions/5130/multiple-keyboard-and-input-languages#answer-60842) for pi is that the keyboard layout setting can't be saved.  I have to go to `etc/default/keyboard` and make the change there. after making the change I can do `ltsp-update-image /opt/ltsp/armhf` it works fine for now. what about the next update or upgrade? is there a way to make this persistent?

gbaman, 2019-08-15:
That type of thing should persist, it is only configuration files that would be stored on the SD card that can get auto overwritten.

I do find it odd that editing the config.txt file via the builtin tool in PiNet didn't work? Can you confirm that it has written the values you entered in the tool, to the config.txt file on the SD card itself?
It by default writes it on the next reboot.

seksitha, 2019-08-15:
Thank you for the prompt response, good to know that persists. I will test that again when I'm back at school tomorrow.

seksitha, 2019-08-16:
@gbaman I spend the whole morning test the custom config. It does not work. what I do is go to cusom-config change `hdmi_group = 9` to see if it overwrite my previous setting. I relaunch Server and reboot Pi serval times. no effect.
And my keyboard setting also does not work after server reload. :(

seksitha, 2019-08-19:
[From doc](http://pinet.org.uk/articles/advanced/custom-config-file.html) said : \
 Once this is complete, simply `reboot all the Raspberry Pis`and they will pull in the new config.txt file, `then reboot to apply it`.\
Does this mean first boot is to load the new config file and second boot to see changed?

Today, I boot up the server again. And the keyboard layout is working again. It seems like changing the Rasbain need both compress and ubuntu reboot. Hmm.

gbaman, 2019-08-19:
The system should notice an updated config.txt file and on reboot, apply the changes and trigger another reboot to apply it.

Can I confirm which version of PiNet your are running (shown at the top of the PiNet software).

As for any changes outside of PiNet, yes you need to recompress the image, as per the documentation (see in red).
http://pinet.org.uk/articles/advanced/editing-outside.html